spacelab-ufsc/interface-board #26: Move RBF pin header from N1 to N2 board

YanAzer, 2020-06-23:
Currently the N1 board (1_Interface_Charge_RBF to be mounted in the -Y plane) has a 2 pin header for RBF switch, for the correct integration in the P-POD it must be moved to the N2 board (2_Interface_JTAG to be mounted in the -X plane).

Source: https://static1.squarespace.com/static/5418c831e4b0fa4ecac1bacd/t/56e9b62337013b6c063a655a/1458157095454/cds_rev13_final2.pdf
[image]
[image]

mgm8, 2020-06-28:
@YanAzer this is not a big issue, as can be seen [here](https://youtu.be/ECmqEms8u_Q?t=302), sometimes the RBF is removed before installing the satellite in the P-POD.

YanAzer, 2020-06-29:
@mgm8 i already did the changes along the week, didn't commited because i didn't want to mix with the history of the N3 board on the recently pull request. Doing this change actually helped since i needed to move the picoblade headers (#27) and somewhat distributed the weight on both PCBs.

commit 8d030a76 (2020-06-29): Semi USB: layout: Removed RBF pin header from N1 board and moved to N2 board PCB layout, fixed fiducials and picoblade headers positioning closes #26 #27 #28
closed 2020-08-08 by YanAzer